arduino-libraries/SD #12: SD example CardInfo doesn't handle large SD cards

agdl, 2016-07-12:
_From @karplus on December 31, 2012 2:39_

I tried reading a 32Gbyte card with CardInfo, and got a ludicrous result for the size.  The problem is that a 32-bit number can't hold the number of bytes on the card.

Replacing the bytes and kbytes printing with the following code fixed the problem, though only for cards up to a Terabyte:

  if (volumesize < (1<<(32-9)))
  {	uint32_t num_bytes = volumesize * 512;                            // SD card blocks are always 512 bytes
  	Serial.print("Volume size (bytes) : ");
  	Serial.println(num_bytes);
  }
  Serial.print("Volume size (Kbytes): ");
  volumesize /= 2;
  Serial.println(volumesize);

_Copied from original issue: arduino/Arduino#1194_

agdl, 2016-07-12:
_From @ceotto on June 23, 2016 21:0_

as ...(strange?) as it can be ... today, in june 2016, the CardInfo example still doesn't work.

Here a piece of code starting to count numbers in Kb not in byte ...

```
  uint32_t  volumesize = (uint32_t)volume.blocksPerCluster() * (uint32_t)volume.clusterCount() / 2UL ;    // clusters are collections of blocks, SD card blocks are always 512 bytes
  Serial.print("Volume size (Kb):\t");   Serial.println(volumesize);
  Serial.print("Volume size (Mb):\t");   volumesize /= 1024; Serial.println(volumesize);
  Serial.print("Volume size (Gb):\t");   volumesize /= 1024; Serial.println(volumesize);  

```
Please: enter this four lines above, and add this few one:
```
  Serial.print("Clusters:\t\t\t");                Serial.println(volume.clusterCount());
  Serial.print("Blocks (512b) x Cluster:\t");     Serial.println(volume.blocksPerCluster());
  Serial.println();

  Serial.print("Total Blocks:\t\t");                
  Serial.println(volume.blocksPerCluster()*volume.clusterCount());
```

So you can have this kind of report:
```
Initializing SD card...
Wiring is correct and a card is present.

Card type: SDHC

Volume type is FAT32

Clusters:			3869184
Blocks (512b) x Cluster:	4

Total Blocks:		15476736
Volume size (Kb):	7738368
Volume size (Mb):	7557
Volume size (Gb):	7

Files found on the card (name, date and size in bytes): 

```

cmaglie, 2017-11-14:
@ceotto 
thanks for the feedback!

commit 8cd6fef1 (2017-11-14): Improved CardInfo example output

Fix #12
closed 2017-11-14 by cmaglie