arduino-libraries/SD #14: SD card initialization fails after SD power down/new init sequence [imported]

agdl, 2016-07-12:
_From @cmaglie on November 15, 2012 19:5_

This is [Issue 1103](http://code.google.com/p/arduino/issues/detail?id=1103) moved from a Google Code project.
Added by 2012-11-07T18:50:46.000Z by [fakuf...@gmail.com](http://code.google.com/u/113444242026466014153/).
Please review that bug for more context and additional comments, but update this bug.

Original labels: Type-Defect, Priority-Medium

### Original description

**What steps will reproduce the problem?**
1. Init SD card
2. Turn SD card off (by cutting power to SD card, but also to all other SD input pins)
3. Turn SD card on
4. Initialize SD card again

**What is the expected output? What do you see instead?**

The expected output would be to have the SD card re-initialize itself. Instead, the second initialization fails.

I have tracked the failure cause to SDFile.openRoot method that return failure if the root had been already opened.

**What version of the Arduino software are you using? On what operating**
**system?  Which Arduino board are you using?**

Arduino IDE 1.0.1
Mac OS X
Custom board

**Please provide any additional information below.**

A fix can be implemented, for example by calling a method that resets the state of the SDClass object in the initialization routine, making sure we start from a blank state.

I am willing to implement the fix, but just want to make sure I implement it the way people would like it.

_Copied from original issue: arduino/Arduino#1103_

SodaqMoja, 2017-10-31:
This is probably related to PR #37 
The caller can first call the end() function before doing init() again.

cmaglie, 2017-11-14:
Fixed by #37 and #38

closed 2017-11-14 by cmaglie