arduino-libraries/SD #5: Change SD Card

agdl, 2016-07-12:
_From @nemeier on August 1, 2015 10:26_

In some of my Projects I need to change the SD card. There is a simple allow this. I just require a single line in the SD.c library.
```
boolean SDClass::begin(uint8_t csPin) {
  /*
    Performs the initialisation required by the sdfatlib library.
    Return true if initialization succeeds, false otherwise.
   */
// ----> Line Added
  if (root.isOpen()) root.close();      // allows repeated calls
// <--- Line Added
  return card.init(SPI_HALF_SPEED, csPin) &&
         volume.init(card) &&
         root.openRoot(volume);
}
```
After changing the SD card all you need is to reinitialize the sd card.

Will properly solve ISSUE #1103

_Copied from original issue: arduino/Arduino#3607_

agdl, 2016-07-12:
_From @dzzie on December 27, 2015 20:44_

this fixed an issue i was having thanks!

agdl, 2016-07-12:
_From @bborncr on March 18, 2016 20:15_

This fix worked for me also. Is there any possible problem that could result from this patch?

tomjennings, 2017-08-06:
i was about to report this same fix for this same problem. i have an automotive data logger (Arduino Mega 2560, etc) that writes only, no reads. this fix was sufficient to allow using write-error to detect card removal, and begin-success to detect card insertion. (a state machine does begin-retry every 5 seconds, etc). i've not found any other side effects. as above the fix was if (root.isOpen()) root.close(); in both begin()s.

cmaglie, 2017-11-14:
Fixed by #38

closed 2017-11-14 by cmaglie