arduino-libraries/SD #31: architectures in library.properties inaccurate

TheNitek, 2017-03-11:
library.properties states architectures=* which makes my Arduino IDE choose it over the SD lib included in the esp8266 board package.
This would not be a problem, if Sd2PinMap.h did not contain
`#error Architecture or board not supported.`
which breaks my build. So I guess the architectures declaration should be more specific.

cmaglie, 2017-11-23:
I've added the definitions https://github.com/arduino-libraries/SD/commit/9543f55fdb08cce6d4ac5798d3b6b7347981a03b

BTW I think that the SD lib provided by the ESP8266 core should have the `architecture` field set as well https://github.com/esp8266/Arduino/blob/7b09ae5f69f8dccb5152d128c8484ec9a18adb11/libraries/SD/library.properties#L9
I'll open an issue over there.

closed 2017-11-23 by cmaglie

Dravon5X, 2017-12-04:
this fix actually produce a bug, that every one user of arudino IDE is receiving notification after IDE start that update is available for SD library, but there is nothing to update. SD library is also always under Updateable libraries. Please revert and find a different way.

commit 9543f55f (2017-11-23): Added architecture specific definitions in library.properties

Fix #31