arduino-libraries/SD #34: SD Library not useable for MKR1000

arduz, 2017-08-30:
Why this library isn't useable for new processors such SAMD21 Cortex-M0+? If I include this library the compiler get an error: The Sd2PinMap.h contains no entrys for this Controller. I did expect, that all new boards, which are add with the board manager, could used with all librarys in the standard IDE, such SD library. Is there any idea how i can use an MKR1000 board with the SD library?

In file included from D:\Arduino\libraries\SD/utility/Sd2Card.h:26:0,

                 from D:\Arduino\libraries\SD/utility/SdFat.h:27,

                 from D:\Arduino\libraries\SD/SD.h:20,

                 from D:\Arduino\Sketches\gpsLogger3\gpsLogger3.ino:6:

D:\Arduino\libraries\SD/utility/Sd2PinMap.h:284:5: error: 'DDRD' was not declared in this scope

per1234, 2017-08-30:
It's somewhat unclear what you're saying. That code is wrapped in a preprocessor conditional that should make it never be seen by the compiler when compiling for MKR1000 (https://github.com/arduino-libraries/SD/blob/1.1.1/src/utility/Sd2PinMap.h#L260). In fact, that code should only be compiled if you have an ATmega644 or ATmega644P based board selected. Did you modify the library?

Please provide a complete, minimal test sketch to reproduce the issue.

arduz, 2017-08-31:
You are right, this failure come from preprocessor.
My failure was: I have installed the new Version (Arduino 1.8.3) oven an older  Version, though the SD-library was not overwritten (??), i have use a older Version of this library.
Now I have installed the newest Version (1.8.5) an all was fine. My sketch is running.

Thanks for your help.

closed 2017-08-31 by arduz