arduino-libraries/SD #45: FILE_WRITE does not work properly since v1.2.0

Nightenom, 2017-11-26:
Command: `File file = SD.open(target, FILE_WRITE);`
How it looked like (version 1.1.1):
```
Reading file "/logs/log3.txt":
[26.11.17][12:59:03][4893] S: GSM: ready without restart
[26.11.17][12:59:08][9526] I: GSM: IP: "100.90.60.208"
[26.11.17][12:59:17][18051] S: RTC: synced
[26.11.17][12:59:17][18077] Boot successful

Done!
```
How it looks like (version 1.2.0):
```
Reading file "/logs/log0.txt":
[01.01.04][00:47:39][60621] Boot successful
86.142.157"

Done!
Reading file "/logs/log1.txt":
[01.01.04][00:52:55][32144] I: GSM: IP: "100.90.60.208"

Done!
Reading file "/logs/log2.txt":
[26.11.17][12:40:57][18082] Boot successful
0.60.208"

Done!
```

closed 2017-11-28 by cmaglie

cmaglie, 2017-11-28:
@Nightenom 
I've pushed a release 1.2.1, it should be available in a couple of minutes, please let me know if this fix the issue, and thanks again for the bug report!

Nightenom, 2017-12-02:
Thanks, I will check it next weekend

commit 38cd35bb (2017-11-28): Fixed regression on FILE_WRITE

Fix #45