arduino-libraries/SD #46: SD library always ask to update itself

Dravon5X, 2017-12-04:
this fix
https://github.com/arduino-libraries/SD/issues/31
https://forum.arduino.cc/index.php?topic=515050.0

actually produce a bug, that every one user of arudino IDE is receiving notification after IDE start that update is available for SD library, but there is nothing to update. SD library is also always under Updateable libraries. Please revert and find a different way.

per1234, 2017-12-04:
It actually just exposes a long-standing bug: https://github.com/arduino/Arduino/issues/4189.

Library Manager should be able to correctly handle a change in the `architectures` value so the solution is to fix the underlying bug, rather than to bandaid over it by reverting https://github.com/arduino-libraries/SD/commit/9543f55fdb08cce6d4ac5798d3b6b7347981a03b.

I'm inclined to close this as a duplicate of https://github.com/arduino/Arduino/issues/4189 but I'll leave it open just in case might nudge some action on the long neglected https://github.com/arduino/Arduino/issues/4189.

amdkt7, 2018-01-02:
Is there a way to fix this locally? Every time I open a new window my screen will suddenly freeze when the notification pops up, and I lose my place. It's getting really annoying.

cmaglie, 2018-01-11:
This is a bug that surely be fixed in the IDE (and I'm working on it on [this branch](https://github.com/cmaglie/Arduino/tree/fix-libraries-installer)), but in the meantime let's alleviate the pain by restoring the `architecture=*` in the `library.properties`.

closed 2018-01-11 by cmaglie

amdkt7, 2018-01-12:
architecture=* in the library.properties? Is that something that I can change locally? I found the file with the "architecture=*" line, but your comment is unclear to me.Thanks for you help

cmaglie, 2018-01-12:
Just update the library to version 1.2.2 in the library manager.

amdkt7, 2018-01-12:
Okay, just did. Last time I checked a few days ago it was not there. Thanks.

commit afa2b16a (2018-01-11): Restored arch=* in library.properties

Fix #46