arduino-libraries/SD #50: File.seek() in SD library--version 1.2.2 didn't work on M0 Pro using the IDE on macOS

egienvalue, 2018-03-02:
Test platform: Arduino M0 Pro

I try to use the seek() function to move the write pointer to the head of a file, it didn't work. Then I change the version to 1.2.0, and it works with the same code.

cmaglie, 2018-03-05:
Please post a test sketch to show the problem.

agrunte, 2018-03-21:
seek() does not work indeed, if the file is opened with FILE_WRITE. But it works if the file is opened with O_WRITE instead:   file = SD.open("test.txt", O_WRITE);

jmpmscorp, 2018-07-05:
Using version 1.2.0 while my requisites were solved, now, I have had to upgrade to 1.2.2 because it was solved any other problem I had with 1.2.0.

Following @agrunte post and another post I could find on google, I can work with seek() function since now, but I should advice something. 

With this version, it seems that, when you open file with FILE_WRITE = (O_READ | O_WRITE | O_CREAT | O_APPEND), this last O_APPEND force file to write always at last position althougth you use seek() function. If you open file with only O_WRITE, if file doesn't exist previosly on SD, it will never be created and it doesn't be write. So, to work with seek() without problem, you could open file with every clause except O_APPEND. 

`` 
file = SD.open("test.txt", O_READ | O_WRITE | O_CREAT);
``