arduino-libraries/SD #62: Fix example ReadWrite

idealsmurffittrainer, 2018-11-16:
I think there is i some mistake in exaple ReadWrite. In code below is pointless while loop which burdens CPU. I'll change if for while and "while(1);" remove
`if (!SD.begin(4)) {
    Serial.println("initialization failed!");
    while (1);
  }`

deepaklorkhatri007, 2020-02-22:
I agree with you, we should at least change the error message with something as I proposed in https://github.com/arduino-libraries/SD/pull/73

per1234, 2022-04-17:
Thanks for taking the time to submit an issue @idealsmurffittrainer.
This perpetual loop is intentional. Continuing with the sketch would be pointless and confusing to the user after the SD card initialization failed.

closed 2022-04-17 by per1234