arduino-libraries/SD #65: SD Card example sketch "CardInfo"

Bambofy, 2019-06-24:
Hi, using the default CardInfo sketch causes the SD card to be corrupted if power to the Arduino stops. I have resolves this issues on my end by adding:

"root.close();"

to the end of the setup(); function to ensure that the sd card is fully closed, and ready to be ejected.

deepaklorkhatri007, 2020-02-22:
just curious question :)
Does this corrupt the SD card entirely or just the files?

Bambofy, 2020-02-22:
> 
> 
> just a curious question :)
> Does this corrupt the SD card entirely or just the files?

The SD card entirely, it required reformatting to work again :)

deepaklorkhatri007, 2020-02-23:
Thanks for the information @Bambofy :) 

> The SD card entirely, it required reformatting to work again :)

I will try to replicate it.

aentinger, 2020-03-18:
Fixed by PR #75 .

closed 2020-03-18 by aentinger