arduino-libraries/SD #82: GPL or LGPL?

EuphoricCatface, 2020-10-06:
SD library seems to have been changed to GPL when it was pointed out that William Greiman's SdFat library was GPL in Apr 2015. https://github.com/arduino/Arduino/issues/2936

In the meantime, in Oct 2017, William Greiman's library was changed to MIT License. https://github.com/greiman/SdFat/commit/e4190e50a4ed83c73fb046b7fce2694e8cfe2ddb and https://github.com/greiman/SdFat/commit/f2424ef75352cae8415ff9c5b5d7206b5bfc2361

As far as I know, for "the stated INTENT of the arduino folk to allow proprietary software trumps", this library is supposed to be LGPL, since it is not mandatory to keep it GPL anymore. https://forum.arduino.cc/index.php?topic=199627.0 (Sorry for not having clearer source. Correct me if I'm wrong.)

jockm, 2021-02-11:
I think it is rather wrong that you have to dig into the libraries to actually see that this library is under a different license than the others in the official arduino examples.

Aside from ideology I see no reason not to go to LGPL.  It's fine to keep it as GPL, but it would be nice to have an official statement of the rational

PaulStoffregen, 2021-02-12:
If anyone from Arduino sees this and wants to update to the latest SdFat library, which brings support for long filenames, cards larger than 32GB, and huge speed improvements, on top of the change from GPL to MIT, I have some code to offer which can help ease the transition.  It's thin SD.h compatibility wrapper.  It also include a File base class abstraction, which allows libraries using files (like AudioZero, JPEGDEC, etc) to access files independent of which library provides them (SdFat, LittleFS, theoretically mbed or others...)

This stuff is on my github account, so anyone can just help themselves to the code.  But when/if Arduino has serious interest in replacing the ancient SD library and a File base class abstraction, I'm willing to help.  All you have to do is ask.