arduino-libraries/SD #95: pathidx may not be initialized

Coopski101, 2021-04-15:
> D:\Program Files\Arduino\libraries\SD\src\SD.cpp: In function 'open.constprop':
> D:\Program Files\Arduino\libraries\SD\src\SD.cpp:462:14: warning: 'pathidx' may be used uninitialized in this function [-Wmaybe-uninitialized]
>      filepath += pathidx;
>               ^
> D:\Program Files\Arduino\libraries\SD\src\SD.cpp:456:9: note: 'pathidx' was declared here
>      int pathidx;
>          ^

This compiler warning no matter what I compile utilizing the library, even examples. I acknowledge a past issue submitted, but it appears to have not been fixed? Or is this an issue on my end?
Thanks

per1234, 2021-04-15:
> I acknowledge a past issue submitted

Is this the one you're referring to?: https://github.com/arduino-libraries/SD/pull/48

Coopski101, 2021-04-15:
> > I acknowledge a past issue submitted
> 
> Is this the one you're referring to?: #48

yes

closed 2022-11-13 by per1234