arduino-libraries/SD #113: Getting the library not working on Arduino Nano Every

arjen8888, 2022-03-16:
Hi,
I have installed the lastest Arduino development environment and I'm using just the SD library that comes as an example. I have the CardInfo example working with an Arduino Uno, but as the title says I'm not getting it working with the Arduino Nano Every.
I already posted some information with pictures on the Arduino Forum and another user suggested to ask here for if there could be a compatibility problem with the Every. Here the link to the topic on the Arduino Forum: [https://forum.arduino.cc/t/sd-card-adapter-on-uno-working-on-nano-every-not-working/967609](https://forum.arduino.cc/t/sd-card-adapter-on-uno-working-on-nano-every-not-working/967609)

I appreciate all tips and suggestions. If any other info is required then please let me know. Thanks for any help.
Arjen.

LukeDS-it, 2022-05-03:
@arjen8888 I found this issue because I'm facing the same problem as you do. Did you find a way to solve it? Is it really code related?

arjen8888, 2022-05-12:
@LukeDS-it  No, I did not solve it yet. A bit frustrated I put the project away. I bought the standard Arduino Nano to give it a try but did not find the time do actually do it. I don't know what the problem is. My latest thoughts were the software because the hardware cannot be more simple. As you probably have read I got it working on the Arduino Uno so you might think by just replacing the board with the Nano Every it should work. Could you please let me know the solution if you find it?