arduino-libraries/SD #120: Unused variable warning

beaver700nh, 2022-06-19:
Hi

Every time I compile my project, I get this warning:
```
/home/pi/Arduino/libraries/SD/src/utility/Sd2Card.cpp:177:16: warning: 'chip_select_asserted' defined but not used [-Wunused-variable]
 static uint8_t chip_select_asserted = 0;
                ^~~~~~~~~~~~~~~~~~~~
```

It would be great if this could be patched so I don't have to deal with a warning that's not mine! Thanks!

per1234, 2022-06-19:
> ```text
> /home/pi/Arduino/libraries/SD/src/utility/Sd2Card.cpp:177
> ```

Hi @beaver700nh. It seems you are using some other version of the SD library than the one from this repository. This repository has that code at line 175 instead of the line 177 indicated by the warning you shared:
https://github.com/arduino-libraries/SD/blob/master/src/utility/Sd2Card.cpp#L175

I also am not able to reproduce this warning when compiling the library from this repository.

So I will close this as off topic for this repository. If you try out the code from this repository and encounter any warnings or other problems, please feel free to report them.

closed 2022-06-19 by per1234

beaver700nh, 2022-06-19:
Hi, I'm compiling with the _highest warning setting_, because Arduino doesn't warn about unused variables normally. The code is on line 177 because I made some modifications that shouldn't affect unused variables.