arduino-libraries/SD #129: Allow using this library on esp boards

gasagna, 2022-11-17:
I am trying to use this library on an esp32 dev board, since the SD library that comes with the arduino-esp32 library (mentioned in some other issues here) is not as feature-complete as this one and leads to random behaviour with some SD cards I have. 

I have attempted to compile my project using this library (and other libs from arduino-esp32, e.g. SPI, ) and I got a

```c++
/Users/gasagna/Documents/Arduino/libraries/SD/src/utility/Sd2PinMap.h:524:4: error: #error Architecture or board not supported.
   #error Architecture or board not supported.
    ^~~~~
```

suggesting that my board is not supported. Well, OK, but ...

I then commented the compile `#error` directive at the end of `Sd2PinMap.h` to get
```c++
#else
  //   #error Architecture or board not supported.
#endif
```
and then initialised the SD card using the appropriate pins for my board, e.g. with
```c++
// sd card pins
#define SD_MISO  2
#define SD_MOSI  15
#define SD_SCLK  14
#define SD_CS    13

SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
SD.begin(SD_CS);
```

The code not only compiles, but works like a charm. 

I wonder what would be the consequences of removing that error directive, or maybe turning it into a more permissive compilation warning.

ark02, 2022-12-21:
I too am trying to use this library with the esp platform and running into the same issues. Why was the compiler directive created for the esp chips? What should I be worried about?

VeloSteve, 2024-03-14:
A reply here would be great.  On Arduino Nano ESP32 I get an error every few hours with the 2.0.13 version of SD.h in the esp32 board package.  Sometimes it even fails right after booting.  SdFat.h is very reliable, but has conflicts with the FS.h library.  If the SD library here would be the best of both worlds I would love to use it.

dzalf, 2025-03-10:
Do you happen to have any updates on this?

I need to use the library with an ESP32 board too

VeloSteve, 2025-03-10:
No update on the software, but I have a couple of comments. First, I have been using SdFat. It works very well for me.

Second, The only solution I found to my reliability problems was to not put two devices on the same SPI. It works fine on single core processors with two cores the approach is not thread safe.  Rather than trying to write two thread safe libraries, I put my display on HSPI and the SD on VSPI.  Now both are very reliable. 

Steve