arduino-libraries/SD #141: New release ?

fpistm, 2023-11-28:
Hi,
do you planned a new release?
Several PR has been merged since the last one. It would be good to perform a new release.
Thanks in advance.

pschatzmann, 2024-07-04:
Do you have any update ?

The obsolete booleans are getting annoying when using any STM32 core because there, boolean are tagged as obsolete and therefore your library gives a lot of compile warnings

Andy2No, 2024-07-04:
@pschatzmann, judging by the 17 pending pull requests, it seems likely that this isn't going to get done soon, if ever.  If you look at the Code page for this repo, there are notes that boolean was replaced by bool, and I only see bool in the files I've looked at.

I haven't tried (I might soon) but it should be simple enough to download the current version (not the release) and install it manually.  On the Code page, click on the Code button (toward top right), then on Download.  That gives you a zip that you may be able to install from directly.  Failing that, extract it and place the folder in your libraries folder (e.g.D:\Documents\Arduino\libraries).  You might want to delete the old one first.

pschatzmann, 2024-07-04:
Thanks for your feed-back: I confirm that would work as a work around. 

However the thing is, that Arduino provides this library  as part of the IDE (in Linux in /home/pschatzmann/.arduino15/libraries/SD) and it leaves a bad feeling that I have to rely on something which is not properly maintained. 

The situation would be slightly different, if I would have decided to install and use this library myself via the library manager.

So I think I rather use SDFAT in this case.

Andy2No, 2024-07-04:
Fair point, @pschatzmann

[SdFat](https://github.com/greiman/SdFat) looks interesting.  I like that it adds exFAT support.