arduino-libraries/SD #146: Inconsistent behavior of FILE_WRITE on Arduino Mega and Nano ESP32

VeloSteve, 2024-01-12:
I have the current version, 1.2.4 installed in the Arduino IDE 2.2.1.  This is the only SD-related library I explicitly have installed.

My application logs data to an SD card by simply calling logFile.print(str) occasionally, opening and closing the file each time since there is a lot of other work that goes on between calls.  This has been working for years with the Mega.  

I just discovered today that on the Nano the logfile is being overwritten from position zero on each call!  The difference is this:

On the Mega

- `SD.open(fileName, FILE_APPEND) ` yields **error: 'FILE_APPEND' was not declared in this scope**
- With FILE_WRITE I get the expected behavior, writing at the end of the file
- This is consistent with the statement **FILE_WRITE (open the file for reading and writing, starting at the end of the file).** [at arduino.cc](https://www.arduino.cc/reference/en/libraries/sd/open/) as referenced in the [README ](https://github.com/arduino-libraries/SD/blob/master/README.adoc)here.

On the Nano ESP32

- FILE_WRITE starts from position zero, failing to append unless I explicitly seek to the end
- FILE_APPEND is accepted, and gives the expected behavior
- This contradicts what is described at arduino.cc

And I just learned something else after some digging.  I could have been using open(fileName, **"a"**) all along like I learned "back in the day".  These "friendly" definitions have done more harm than good in this case.

per1234, 2024-01-13:
Hi @qbox4u. Thanks for your report.

The 3rd party ESP32 boards platform comes with its own bundled version of the SD and SPI libraries, which have been modified specifically for use with the ESP32 microcontroller. That library code is hosted in the repository of that boards platform:

https://github.com/espressif/arduino-esp32/tree/master/libraries/SD

This repository is for the official Arduino SD library, which is not used when compiling for an ESP32 board. It is used when compiling for the Mega board. Since you report that the library works as expected on the Mega, this tells me that the problem is specific to the ESP32 variant of the library an so I will close the issue as off topic for this repository.

If you need assistance with your project, feel free to post on the Arduino Forum. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/

closed 2024-01-13 by per1234

VeloSteve, 2024-01-13:
I agree that there is nothing to change in the SD code base here.  **However**, if Arduino is going to sell both AVR and ESP boards the documentation links need to be updated so that they are not misleading.