arduino-libraries/SD #147: Could not find FAT16/FAT32 partition

victorlysak, 2024-02-16:
trying this library with SanDisk SDHC 32 gb with UNO.

All wiring is correct - checked many times.
Card formatted with GParted with part table msdos, fat32, flags: lba

When running the cardinfo example the terminal, I get:

Initializing SD card...Wiring is correct and a card is present.

Card type:         SDHC
Could not find FAT16/FAT32 partition.
Make sure you've formatted the card

What's the problem?

thanx

per1234, 2024-02-16:
Hi @victorlysak. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2024-02-16 by per1234

victorlysak, 2024-02-16:
not true

tried with SdFat and the output on the same card is:
type any character to start
init time: 73 ms

Card type: SDHC
sdSpecVer: 6.00
HighSpeedMode: false

Manufacturer ID: 0X3
OEM ID: SD
Product: SD32G
Revision: 8.5
Serial number: 0XFCF19506
Manufacturing date: 3/2022

cardSize: 31927.04 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
dataAfterErase: zeros

OCR: 0XC0FF8000

read MBR failed.
SD errorCode: SD_CARD_ERROR_CMD18 = 0XC
SD errorData = 0XF