arduino-libraries/SD #160: CardInfo example doesn't compile on ESP32S3

holla2040, 2024-09-24:
Arduino IDE 2.3.2 compiling for 'UM TinyS3' doesn't compile.

tmp/.arduinoIDE-unsaved2024823-10184-1czkmb8.r1lm/CardInfo/CardInfo.ino:26:1: error: 'Sd2Card' does not name a type
   26 | Sd2Card card;
      | ^~~~~~~
/tmp/.arduinoIDE-unsaved2024823-10184-1czkmb8.r1lm/CardInfo/CardInfo.ino:27:1: error: 'SdVolume' does not name a type
   27 | SdVolume volume;
      | ^~~~~~~~
/tmp/.arduinoIDE-unsaved2024823-10184-1czkmb8.r1lm/CardInfo/CardInfo.ino:28:1: error: 'SdFile' does not name a type; did you mean 'SDFile'?
   28 | SdFile root;
      | ^~~~~~

[image]

per1234, 2024-09-24:
Hi @holla2040. When you compile for an ESP32 board, a different "SD" library than the one hosted in this repository is used:

https://github.com/espressif/arduino-esp32/tree/master/libraries/SD

There are some differences in the API of that library, which is why you can't compile this SD library's example sketches with the ESP32 "SD" library.

You can learn how to use the ESP32 "SD" library by studying the example sketches you will find under the **File > Examples > SD** menu in Arduino IDE.

If you would like assistance using the library, you are welcome to post over on Arduino Forum:

https://forum.arduino.cc/

closed 2024-09-24 by per1234

holla2040, 2024-09-24:
@per1234 thank you for the clarification and the quick response.